7Ji/archrepo #1: More packages?

hbiyik, 2023-08-30:
Hello I am the author of mpp related packages in AUR.

I am using your repo and thanks for that, would you be interested in adding more packages required for rockchip related products.

[linux-radxa-rkbsp5-git](https://aur.archlinux.org/packages/linux-radxa-rkbsp5-git): Required for Rock5a/b
[kodi-nexus-mpp-git](https://aur.archlinux.org/packages/kodi-nexus-mpp-git): Kodi Nexus patched with mpp. (all sub packages as well)
[kodi-matrix-mpp-git](https://aur.archlinux.org/packages/kodi-matrix-mpp-git): Kodi Matrix patched with mpp. (all sub packages as well)

hbiyik, 2023-08-30:
@kwankiu FYI

7Ji, 2023-08-30:
I've pulled these PKGBUILDs: https://github.com/7Ji/archrepo/commit/fc406034eb1b9d805b904b9b311400cc9c22fb53

But holy molly you're using .git kernel source and that was slow to clone, you'll need to wait for several hours before I can start building them

hbiyik, 2023-08-30:
wow, youre super fast, but please dont build linux package yet i noticed that i was building it wrong, and i need to update that.
https://github.com/radxa/kernel/pull/197.

And i also have another concern about the linux package of yours, you are using the libmali firmware as a seperate package

https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=libmali-valhall-g610

But there is a dependency in between them, and it is very hard to tell it. The more the BSP version the kernel increases, the higher it requires the minimum version of the frimware.

I have for rock5 two different versions of kernels, 1st is the prebuilt binary from radxa ([linux-radxa-rkbsp5-bin](https://aur.archlinux.org/packages/linux-radxa-rkbsp5-git)) this is on rkr3.4 bsp version and [linux-radxa-rkbsp5-git](https://aur.archlinux.org/packages/linux-radxa-rkbsp5-git) this is on rkr4.1 version. And the require different versions of the libmali firmware.

Previously i also had a seperate libmali package but for the above reason i have embedded them to the linux package.

hbiyik, 2023-08-30:
i am only pulling the linux .git to calculate the right version in pkgver since the initial package was meant to be used only on aur without the need of future maintenance.

7Ji, 2023-08-30:
The firmware and libmali are both g13p0, same as the DDK version used in OPi kernel, and is what upstream RK hands out to each vendors. There is only one opi5 kernel package and it was maintained by myself so there was no such concern. 

And i don't think it's a good idea to include userspace libraries in a kernel package, you could have a bsp group that pulls in both the kernel and the lib as dep but don't employ them as ome single package.

Also, that libmali package sticks to upstream, not to OPi, it's the latest version available from rk, not opi. If an older version is needed then it should be provided with an embeded version name, just like ffmpeg4.4.

Between your self-build and -bin packages, which kernel DDK versions are they using? Is it that g13p0 is too old that breaks your kernel or new enough that nothing bad happens?

7Ji, 2023-08-30:
So it seems opi is using g13p0: https://github.com/orangepi-xunlong/linux-orangepi/blob/eb1c681e5184e51d8ce1f351559d149d17f48b57/drivers/gpu/arm/bifrost/Kbuild#L74

radax 4.1 uses g17p0: https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr4.1/drivers/gpu/arm/bifrost/Kbuild#L72

radax 3.4 uses g13p0: https://github.com/radxa/kernel/blob/c4c032bbf7599857ba5f42506b64cb6d0b3e88d4/drivers/gpu/arm/bifrost/Kbuild#L74

latest from rk is g18p0:
https://github.com/JeffyCN/mirrors/commit/ca33693a03b2782edc237d1d3b786f94849bed7d

I guess I'll maintain the firmware package seperately from the libmali package, being latest g18p0, then have a series of  dedicated suffixed firmware package for these kernels needing old firmware

hbiyik, 2023-08-30:
i was just writing the same, but do you know if the later versions of 

1) firmware
2) user space drivers

would work with kernel requiring an older version. I think my experience was the other way around, i had the lower version of firmware installed but kernel required a higher version.

If that would work, both 3 kernels can share the same external firmware / userspace driver..

7Ji, 2023-08-30:
In fact I am using g18p0 on my opi5, after encountering some glitches with g13p0. So later firmware + older kernel DDK do work

hbiyik, 2023-08-30:
ok then lets use this way, i will also test rkr34 and rkr41 for radxa using your packages with g18 firmware and userpace libs

hbiyik, 2023-08-30:
one another topic:

do you think it would make sense to include a runner script to your libmali package so users wont have to deal with LD_LIBRARY_PATH and LD_PRELOAD things.

https://aur.archlinux.org/cgit/aur.git/tree/libmali?h=linux-radxa-rkbsp5-git

7Ji, 2023-08-30:
No it won't be g18 library. For whatever reason rk only updated the firmware in their repo but no the library. So the latest is g18 firmware + g13 libraries. My mentioned test case was g13 kernel DDK + g18 firmware + g13 library. I would update the package so it would pack g18 firmware and g13 library.

The runner script could be added for each split package.

hbiyik, 2023-08-30:
ah i see, i will also check practically tonight how this will work in radxa kernel, i expect it to work.

Note the package would depends on gl4es for this script to work. It is also wrapping the OpenGL libs to gl4es.

7Ji, 2023-08-30:
Mali libraries and firmware are now splitted into two PKGBUILDs:
 - https://aur.archlinux.org/pkgbase/libmali-valhall-g610
 - https://aur.archlinux.org/packages/mali-valhall-g610-firmware

The wrapper script is provided as part of https://aur.archlinux.org/packages/libmali-valhall-g610-base as a multi-call script: https://aur.archlinux.org/cgit/aur.git/tree/libmali-wrapper?h=libmali-valhall-g610 , each split package has its own symlink pointing to it, e.g. `libmaliw` owned by `-wayland-gbm`. Calling `libmali` directly does what your old script does.

hbiyik, 2023-09-01:
@7ji,kodi is ok, but linux packages need some extra work still, radxa have changed the whole repository the config and now they have introduced dkms drivers so it will take some time. sorry.

hbiyik, 2023-09-01:
@7Ji now the kernel is ready to be built.

7Ji, 2023-09-02:
All packages requested added. Took a while to re-build all packages as I switched to a new building framework, but future updates will take less effort.

closed 2023-09-02 by 7Ji