7Ji/archrepo #5: python dependency of rockhip kernel

hbiyik, 2023-09-04:
https://github.com/rockchip-linux/kernel/commits/develop-4.4/scripts/gcc-wrapper.py
https://github.com/rockchip-linux/kernel/commits/develop-5.10-rt53/scripts/clang-wrapper.py

https://github.com/torvalds/linux/tree/master/scripts

It seems like rockchip kernel has been using python as the scripting language. This was not a visible problem when building the kernel because makedepends dtc was already depending on python. However when building a kernel module out of tree (ie with dkms or manually out of tree building), there is no "depends" that can satisfy python and it will fail.

On most systems python is for whatever reason is installed so time to time it is possible the issue could not be seen.

I figured out the best solution to overcome this is to add "python" to "depends" of linux-headers package which is based on rockchip kernel. I saw you also have a orangepi kernel i think this should also impact you. What do you think?

7Ji, 2023-09-04:
Indeed, compiling anything against the headers in tree needs Python. It was ignored before as dtc, one of the makedepends of the linux packages itself, introduced python as its dependency, and I didn't really have a reason to build a DKMS module.

About the dep, traditionally the headers package shouldn't depend anything to work. There is a blurry line about whether the headers work, DKMS calling the in-tree scripts being a upper layer example, but there are packages that only need the headers themselves, like https://gitlab.archlinux.org/archlinux/packaging/packages/openssh/-/blob/main/PKGBUILD?ref_type=heads#L26

And according to PKGBUILD.5, only packages needed to make one package work at all should be added to its depends, otherwise they should be listed as optdepends. So I'd rather put Python as a optdepend.

hbiyik, 2023-09-04:
> Indeed, compiling anything against the headers in tree needs Python.

Is it the same case for mainline out of tree module build cases? I saw the need for gcc_wrapper.py & clang_wrapper.py for vendor kernel. Those file are not in the mainline.

> So I'd rather put Python as a optdepend.

In that case how can a dkms package or the user who installs it or the tool (ie pacman) know that they should also install the optdepends of the headers?

7Ji, 2023-09-04:
> In that case how can a dkms package or the user who installs it or the tool (ie pacman) know that they should also install the optdepends of the headers?

When someone installs that package, pacman would tell them the optdepends and the reasons.

hbiyik, 2023-09-04:
ok understood, 
> Those file are not in the mainline.
i checked this, and yes mainline does not require python to build modules, this is only a rockchip kernel thing.

hbiyik, 2023-09-05:
https://gitlab.archlinux.org/archlinux/packaging/packages/linux/-/blob/main/PKGBUILD?ref_type=heads#L128

Ok reopenning, clearly there is a dependency since 5.1.2 in archlinux repo do you know why this pahole is there?

7Ji, 2023-09-05:
https://bugs.archlinux.org/task/69654

So likewise we should add python as dep.

But that makes me wonder whether Python is the only dep needed. Because if you want everything to work in /usr/lib/modules/whatever/build, then there are al least also other scripting languages needed for those scripts...

hbiyik, 2023-09-05:
Not everything under that, only a certain path where the module compiling happens in the Makefile, and that only requires python, not pahole or nothing else, i have tested and verified 5 different dkms modules as well to make sure..

I have updated rkbsp kernel headers to require python with depends. It will be fixed on the next release of your binary repo.

closed 2023-09-06 by 7Ji

7Ji, 2023-10-18:
As I'm implementing a non-bash PKGBUILD parser and I went through the internal logic of how each split package's attribute is parsed by makepkg, I found out your https://aur.archlinux.org/packages/linux-radxa-rkbsp5-git has invalid split package functions. At the end of the PKGBUILD you've written it like this:
```
for _p in ${pkgname[@]}; do
  eval "package_${_p}() {
    _package${_p#${_kernel}}
  }"
done
```
In my/others' kernel packages, we wrote like this:
```
for _p in "${pkgname[@]}"; do
  eval "package_$_p() {
    $(declare -f "_package${_p#$pkgbase}")
    _package${_p#$pkgbase}
  }"
done
```
The reason we used `declare -f` is to embed the actual variables and the logic textural-ly into the result package_whatever function, as makepkg later relies on `declare -f package_whatever`  to extract those from split packages. With your style that results in the following chunk when makepkg wants to extract those attributes:
```
package_linux-radxa-rkbsp5-git () 
{ 
    _package
}
```
And thus makepkg would be unable to read the depends/provides of split packages.

You can verify that by your [SRCINFO](https://aur.archlinux.org/cgit/aur.git/tree/.SRCINFO?h=linux-radxa-rkbsp5-git), the split packages' attributes are missing, your added python dependency is nowhere to be found.