7Ji/archrepo #8: Add packages for Khadas Edge 2?

kwankiu, 2023-10-03:
Hello, I am using your repo for https://github.com/kwankiu/archlinux-installer-rock5 and thanks for that.
As I am receiving request to add support for the Khadas Edge 2, would you be interested in adding the required kernel and firmware for the Edge 2?

Kernel for the Edge 2 : https://github.com/BredOS/sbc-pkgbuilds/tree/main/linux-edge2
Wifi firmware for the Edge 2 : https://github.com/BredOS/sbc-pkgbuilds/tree/main/ap6275p-firmware

@DarkevilPT FYI

7Ji, 2023-10-03:
I have a few questions:

 - Can you share the link to the request you're receiving? I want to make sure these two are the only two needed.
 - Are these two PKGBUILDs the most upstream ones? And you're sure there's no pre-built repo for them?
   I would prefer packages maintained alone from e.g. AUR,, then those maintained alone from github, e.g.  https://github.com/7Ji-PKGBUILDs , then from distro repos. 
   The order is also from least likely to have pre-built packages to most likely to have pre-built packages. And these two are from BredOS's repos so they should already provide the pre-built packages.

kwankiu, 2023-10-03:
I think that BredOS has its own repo, but I am worried about using both their repo and your repo at the same time, as there may be packages that conflicts, and I think that their repo is less up-to-date. So I am just thinking whether it would be a good idea to have Edge 2's kernel and firmware on your repo instead.

kwankiu, 2023-10-03:
> * Can you share the link to the request you're receiving?

I got request for adding Edge 2 from Discord direct messaging, they hope that I could bring Arch Linux to the Edge 2 (instead of the current only option - BredOS)

kwankiu, 2023-10-03:
https://docs.khadas.com/products/sbc/edge2/development/linux/linux-source-code points to https://github.com/khadas/linux/tree/khadas-edges-5.10.y-release-v1.0.0

So yes, the PKGBUILDs seems to be using the most upstream ones
https://github.com/BredOS/sbc-pkgbuilds/blob/main/linux-edge2/PKGBUILD

7Ji, 2023-10-03:
They do have conflicts, most notably they're hosting my `linux-aarch64-orangepi5` package as `linux-orangepi5` and it is the version I pushed to AUR long time ago: https://github.com/BredOS/sbc-pkgbuilds/blob/main/linux-orangepi5/PKGBUILD

But repos won't conflict on providing packages, even if they have the same packages. You can always prefix the package name with repo to force Pacman to choose which one to use, e.g. 7Ji/linux-aarch64-orangepi5, if not forced they it would choose the one from the one defined earlier in the config.

If we really are adding this then I want this to be:
  - Maintained elsewhere, not as a part of the BredOS repo.  It would suck if package fixes need to pass through them first considering how out-of-date they're. You could open a repo, or let me open one under 7Ji-PKGBUILDs are invite you as maintainer.
  - Verified to work on actual hardware. Bear in mind every package on this repo needs to be built on my OPI5 Plus + OPi 5 combo. The periodical full rebuild is taking more and more time and closing to 24h due to more packages being added. Even if I don't mind the time & power, I can only test them on OPi5, not on other hardware. So I don't want to add packages that don't even work.

7Ji, 2023-10-04:
Another thing that makes it hard to add these packages was that the builder lacks the capability to handle PKGBUILDs from git repos that do not store PKGBUILD at the root tree on master branch. After some changes this is now possible. You can build those packages with the following config.yaml:
```
pkgbuilds:
  linux-edge2:
    url: https://github.com/BredOS/sbc-pkgbuilds.git
    branch: main
    subtree: linux-edge2
  ap6275p-firmware:
    url: https://github.com/BredOS/sbc-pkgbuilds.git
    branch: main
    subtree: ap6275p-firmware
```
To use the builder:
```
git clone https://github.com/7Ji/arch_repo_builder
mkdir repo
cd arch_repo_builder
cargo build --release && strip target/release/arch_repo_builder -o ../repo/arch_repo_builder
cd ../repo
echo '
pkgbuilds:
  linux-edge2:
    url: https://github.com/BredOS/sbc-pkgbuilds.git
    branch: main
    subtree: linux-edge2
  ap6275p-firmware:
    url: https://github.com/BredOS/sbc-pkgbuilds.git
    branch: main
    subtree: ap6275p-firmware' > config.yaml
sudo ./arch_repo_builder
```
The result packages would be stored under `pkgs/latest`

Please report if these packages work on actual hardware.

kwankiu, 2023-10-05:
@andrebraga Could you help testing with your board?

7Ji, 2023-10-07:
The firmware package is buildable:
[ap6275p-firmware-20231006-1-aarch64.pkg.tar.gz](https://github.com/7Ji/archrepo/files/12836689/ap6275p-firmware-20231006-1-aarch64.pkg.tar.gz)

The kernel package, however, is clearly broken and not buildable, it needs to be fixed and maintained outside of the bred repo:
```
==> Making package: linux-edge2 5.10.66-1 (Sat Oct  7 03:34:29 2023)
==> WARNING: Skipping dependency checks.
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
/home/nomad7ji/build/repo/build/linux-edge2/PKGBUILD: line 40: pkgver: No such file or directory
Makefile:669: include/config/auto.conf: No such file or directory
==> ERROR: A failure occurred in build().
    Aborting...
```

(also I doubt if that was even buildable in the first place, it is not broken due to source update, it was a broken PKGBUILD to begin with, maybe bred maintainers didn't push their actual PKGBUILDs?)

Someone with edge2 needs to fork their kernel package out, and test & debug to fix it up, before it is possible to add to my repo.

closed 2023-10-08 by 7Ji