7Ji/archrepo #10: texinfo > breaks  `ffmpeg-mpp` build

7Ji, 2023-11-07:
Since the update of `texinfo` ~2 weeks ago, the `ffmpeg-mpp` package could not be built due to some configuration format change, the same applies to `ffmpeg` regardless of the distro, as long as they got latest `texinfo`. Arch `ffmpeg` is not 'affected' because it's not been rebuilt yet.

Debian devs have written a patch for their ffmpeg package which fixes the problem for them: https://salsa.debian.org/multimedia-team/ffmpeg/-/blob/debian/master/debian/patches/0005-Patch-up-t2h-for-texinfo-7.1.patch . This repo however does not maintain the PKGBUILD for `ffmpeg-mpp` so that could not be applied yet.

@hbiyik would you please check if that patch could be added to your PKGBUILD?

7Ji, 2023-11-07:
The build failure log:
```
GENTEXI doc/avoptions_codec.texi
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
HTML    doc/ffmpeg.html
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
HTML    doc/ffplay.html
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
makeinfo: error parsing ./doc/t2h.pm: Undefined subroutine &Texinfo::Config::set_from_init_file called at ./doc/t2h.pm line 24.
make: *** [doc/Makefile:70: doc/ffmpeg.html] Error 1
make: *** Waiting for unfinished jobs....
makeinfo: error parsing ./doc/t2h.pm: Undefined subroutine &Texinfo::Config::set_from_init_file called at ./doc/t2h.pm line 24.
make: *** [doc/Makefile:70: doc/ffplay.html] Error 1
==> ERROR: A failure occurred in build().
    Aborting...
```

7Ji, 2023-11-07:
@JFLim1 

The attachment is the last successful build before `texinfo` update
[ffmpeg-mpp-2.6.0-9-aarch64.tar.gz](https://github.com/7Ji/archrepo/files/13285128/ffmpeg-mpp-2.6.0-9-aarch64.tar.gz)

JFLim1, 2023-11-08:
> The attachment is the last successful build before `texinfo` update [ffmpeg-mpp-2.6.0-9-aarch64.tar.gz](https://github.com/7Ji/archrepo/files/13285128/ffmpeg-mpp-2.6.0-9-aarch64.tar.gz)

Hi @7ji,
Thank you.

hbiyik, 2023-11-08:
@7Ji  ok thanks for reporting, i think this should go away when documentation is disabled on configure. I dont want to patch ffmpeg. I will update tonight.

hbiyik, 2023-11-09:
both ffmpeg-mpp & ffmpeg4.4-mpp updated, should be fixed now

7Ji, 2023-11-09:
Nice, `ffmpeg-mpp` was successfully rebuilt. I'll close the issue after the whole build circle is finished and the package files are uploaded.

```
==> Finished making: ffmpeg-mpp 2:6.0-10 (Thu Nov  9 13:33:04 2023)
[2023-11-09T13:33:04Z INFO  arch_repo_builder::build::builder] End of log for building 'ffmpeg-mpp'
[2023-11-09T13:33:04Z INFO  arch_repo_builder::pkgbuild] Finishing building 'ffmpeg-mpp-bf0797549552cdf942a4058961de88cf605d4b91-dc005b89fc78f5fb-6.0'
[2023-11-09T13:33:04Z INFO  arch_repo_builder::pkgbuild] Finished building 'ffmpeg-mpp-bf0797549552cdf942a4058961de88cf605d4b91-dc005b89fc78f5fb-6.0'
[2023-11-09T13:33:04Z INFO  arch_repo_builder::build::builder] Successfully built 'ffmpeg-mpp'
```

closed 2023-11-09 by 7Ji

JFLim1, 2023-11-11:
Hi @hbiyik,

Installed ffmpeg-mpp successfully but mpv would not stream any video -- just a blue screen but with sound. Error:
```
[jfl@jfl-opi5plus ~]$ mpv --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 () (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 () (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Using hardware decoding (rkmpp).
AO: [pipewire] 48000Hz 5.1 6ch floatp
VO: [gpu] 3840x2160 drm_prime[nv12]
[vo/gpu/drmprime] Mapped surface with format 'nv12' has unexpected number of planes. (0 layers and 0 planes, but expected 2 planes)
[vo/gpu/drmprime] mapping DRM dmabuf failed
[vo/gpu] Mapping hardware decoded surface failed.
AV: 00:00:00 / 00:10:34 (0%) A-V:  0.000 Cache: 0.4s/308KB
```