7Ji/archrepo #11: Warning to package maintainers: ongoing Deletion Request spam on AUR

7Ji, 2023-11-13:
An AUR user (not dev, not TU, not assuming Package Maintainer role, JUST A USER) MarsSeed is currently going rampage on AUR to file Deletion Request to a LOT of packages, affecting me starting from PRQ#50220, including PRQ#50222, PRQ#50980, correponding each to `linux-firmware-amlogic-ophub`, `linux-firmware-orangepi`, `chromium-mpp` I'm building for the repo.

He also files a lot of Deletion Request against pacakges maintained by @hbiyik , `ffmpeg-mpp`, `ffmpeg4.4-mpp`, `librga-multi`, `kodi-matrix-mpp-git`, etc are all currently being affected.

You could check the `aur-request` mailing list that almost half of the ongoing Deletion Requests are files by him: https://lists.archlinux.org/hyperkitty/list/aur-requests@lists.archlinux.org/

We NEED to take action

7Ji, 2023-11-13:
I think we need to backup the PKGBUILDs in case they're deleted

PKGBUILDs I'm maintaining on AUR:
```
ampart
ampart-git
beep-sudoable
chromium-mpp
git-mirrorer
libmali-valhall-g610
libv4l-rkmpp-git
linux-aarch64-flippy
linux-aarch64-flippy-bin
linux-aarch64-orangepi5
linux-firmware-amlogic-ophub
linux-firmware-amlogic-ophub-git
linux-firmware-orangepi
linux-firmware-orangepi-git
mali-valhall-g610-firmware
openvfd
pacman-beep-hook
rtl8189fs-dkms-git
uboot-legacy-initrd-hooks
usb2host
v4l-utils-mpp
yaopenvfd
```

PKGBUILDs hbiyik is maintaining:
```
8852be-dkms-git
8852bu-dkms-git
dri2to3-git
droidmote-bin
ffmpeg4.4-mpp
ffmpeg-mpp
gl4es-git
kodi-matrix-addon-inputstream-adaptive-git
kodi-matrix-addon-pvr-iptvsimple-git
kodi-matrix-mpp-git
kodi-nexus-addon-inputstream-adaptive-git
kodi-nexus-mpp-git
librga-multi
linux-radxa-rkbsp5-bin
linux-radxa-rkbsp5-git
mpp-git
r8125-dkms-git
```

7Ji, 2023-11-13:
Just WTF is he doing? Spying on AUR all day?
```
MOST ACTIVE POSTERS
notify＠aur.archlinux.org
3189 posts
Marcell Meszaros
71 posts
Fabio Loli
6 posts
sL1pKn07 SpinFlo
5 posts
Muflone
4 posts
```

7Ji, 2023-11-13:
I've mirrored all our PKGBUILDs to https://github.com/orgs/7Ji-PKGBUILDs/repositories , and invited @hbiyik as org member of @7Ji-PKGBUILDs. In case his requests are accepted we can maintain these packages there.

hbiyik, 2023-11-13:
Some people really needs to get a life, seriously

hbiyik, 2023-11-13:
https://bbs.archlinux.org/viewtopic.php?pid=2131460#p2131460

hbiyik, 2023-11-13:
Right choice no need to argue if they go fascist on their desicions

kyak, 2023-11-13:
:angry: I'm with you guys

hbiyik, 2023-11-13:
@kyak thanks
@7Ji i had just sat down on PC to mode all the PKGbuilds to the github but you have already done this as well, thanks for your work :)

7Ji, 2023-11-14:
@Integral-Tech 

Sorry for the ping, but I fell the need to let you know that a lot of my aarch64-specific packages were deleted from AUR (also others' packages, but I can't track the full list for them) as the result of the ongoing AUR purging.

Among them, based on a simple search, at least `alarmcn/linux-aarch64-orangepi5`, `alarmcn/mali-valhall-g610-firmware`  are affected as they lose the AUR upstreams to update from. I don't know how lilac bot would handle this, but if it needs a new upstream, these packages were migrated to https://github.com/7Ji-PKGBUILDs before the deletion.

Integral-Tech, 2023-11-14:
> @Integral-Tech
>
> Sorry for the ping, but I fell the need to let you know that a lot of my aarch64-specific packages were deleted from AUR (also others' packages, but I can't track the full list for [...]

I removed ```alarmcn/linux-aarch64-orangepi5``` & ```alarmcn/mali-valhall-g610-firmware``` from archlinuxcn as you had created a pacman repo for these packages.

christian-heusel, 2023-11-16:
> I think we need to backup the PKGBUILDs in case they're deleted

Just to note here, PKGBUILDs are not deleted once a deletion request is accepted, they are just unlisted (the underlying git repo is still intact) :blush:

7Ji, 2023-11-16:
> > I think we need to backup the PKGBUILDs in case they're deleted
> 
> Just to note here, PKGBUILDs are not deleted once a deletion request is accepted, they are just unlisted (the underlying git repo is still intact) 😊

Thank you for the note.

Yes I know that. The main reason I migrated them to a new place is for new commits to happen. With new commits pushed to AUR, the repo would again be listed and that would mostly be considered an objection to previous PM decisions which is really bad and I don't want to risk that.

Also another reason is that my builder used AUR API to determine if package source is updated to save network access if the URL matches AUR, before actually using libgit2 to update them. With them unlisted the builder could no longer do that time-saving trick, and would cause much more direct acesses to AUR server.

Thank you still for the note. I think it's my fault to announce the situation to others contributing to my repo without telling them the repos are still there.

7Ji, 2024-01-13:
https://lists.archlinux.org/hyperkitty/list/aur-general@lists.archlinux.org/thread/GOOJIO56E2GGTBPF4FHTYXSUKPBJ3BIQ/

FYI, MarsSeed's account was suspended.

hbiyik, 2024-01-14:
this whole and the previous ones, him in the center or in the circle, right or wrong, are just a waste of time without any particular benefit.

I hope they can focus on productive things.