7Ji/archrepo #12: Could you please add some non-generic packages to your AArch64 repository?

DragonSerW, 2023-11-17:
Hello 7Ji!
Could you please add such packages:
```bash
sudo pacman -S devilutionx devilutionx-voices fheroes2 gzdoom rustdesk-bin vcmi ventoy-bin woeusb-ng xonotic
```
Need them badly for my Orange Pi 5B.

7Ji, 2023-11-18:
gzdoom: on chaotic AUR for x86_64

rustdesk-bin: on chaotic AUR for x86_64

ventoy-bin: on chaotic/archlinuxcn for x86_64, on arch4edu for aarch64

woeusb-ng: on chaotic/archlinuxcn for x86_64, on archlinuxcn for aarch64

xonotic: on official extra for x86_64, maybe ask official ALARM on why it's not built?

If you want to add packages, open a PR with aarch64.yaml and/or x86_64.yaml modified instead of an issue, we shall discuss there. But the above mentioned packages are discouraged.

7Ji, 2023-11-18:
https://github.com/7Ji/archrepo#contribution

closed 2023-11-18 by 7Ji

DragonSerW, 2023-11-18:
https://github.com/arch4edu/arch4edu, https://github.com/archlinuxcn/repo.
Thank you pal, you are the best! :-)
Solution for issue with adding archlinuxcn key: https://bbs.archlinux.org/viewtopic.php?id=258247

DragonSerW, 2023-11-18:
@7Ji, please tell me, how should I contact `alarm` about the question of Xonotic to be added into https://archlinuxarm.org/? :-)

7Ji, 2023-11-19:
https://archlinuxarm.org/forum/viewforum.php?f=5&sid=898da8cabc8768262b7feca38c80a4cd