7Ji/archrepo #15: Cannot Install FFMPEG-MPP -- Due To Invalid or Corrupted Package (checksum)

JFLim1, 2023-12-14:
```
sudo pacman -S ffmpeg-mpp
resolving dependencies...
looking for conflicting packages...
:: ffmpeg-mpp and ffmpeg are in conflict. Remove ffmpeg? [y/N] y

Packages (12) cjson-1.7.16-1  ffmpeg-2:6.1-2 [removal]  librga-multi-1.9.3.1-1
              librist-0.2.10-1  mbedtls-3.4.1-1  meson-1.3.0-1
              mpp-git-r3429.b4be2b72-1  ninja-1.11.1-3  onevpl-2023.4.0-1
              python-tqdm-4.66.1-2  python-typing_extensions-4.8.0-1
              ffmpeg-mpp-2:6.0-10

Total Download Size:   12.14 MiB
Total Installed Size:  56.77 MiB
Net Upgrade Size:      21.41 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 ffmpeg-mpp-2.6.0...    12.1 MiB  1378 KiB/s 00:09 [######################] 100%
(11/11) checking keys in keyring                   [######################] 100%
(11/11) checking package integrity                 [######################] 100%
:: File /var/cache/pacman/pkg/ffmpeg-mpp-2.6.0-10-aarch64.pkg.tar.zst is corrupted (invalid or corrupted package (checksum)).
Do you want to delete it? [Y/n] Y
error: failed to commit transaction (invalid or corrupted package (checksum))
Errors occurred, no packages were upgraded.
```

7Ji, 2023-12-14:
https://github.com/7Ji/archrepo?tab=readme-ov-file#aggresive-rebuild-warning

JFLim1, 2023-12-14:
Follow the steps given https://github.com/7Ji/archrepo?tab=readme-ov-file#aggresive-rebuild-warning but did not work.

```
sudo pacman -Scc
[sudo] password for jfl: 

Cache directory: /var/cache/pacman/pkg/
:: Do you want to remove ALL files from cache? [y/N] y
removing all files from cache...

Database directory: /var/lib/pacman/
:: Do you want to remove unused repositories? [Y/n] 
removing unused sync repositories...

sudo pacman -S ffmpeg-mpp
resolving dependencies...
looking for conflicting packages...
:: ffmpeg-mpp and ffmpeg are in conflict. Remove ffmpeg? [y/N] y

Packages (12) cjson-1.7.16-1  ffmpeg-2:6.1-2 [removal]  librga-multi-1.9.3.1-1
              librist-0.2.10-1  mbedtls-3.4.1-1  meson-1.3.0-1
              mpp-git-r3429.b4be2b72-1  ninja-1.11.1-3  onevpl-2023.4.0-1
              python-tqdm-4.66.1-2  python-typing_extensions-4.8.0-1
              ffmpeg-mpp-2:6.0-10

Total Download Size:   17.51 MiB
Total Installed Size:  56.77 MiB
Net Upgrade Size:      21.41 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 mbedtls-3.4.1-1-...   980.2 KiB  1077 KiB/s 00:01 [######################] 100%
 onevpl-2023.4.0-...   837.3 KiB   891 KiB/s 00:01 [######################] 100%
 meson-1.3.0-1-any    1989.4 KiB  1907 KiB/s 00:01 [######################] 100%
 ninja-1.11.1-3-a...   144.8 KiB   837 KiB/s 00:00 [######################] 100%
 mpp-git-r3429.b4...  1152.6 KiB   977 KiB/s 00:01 [######################] 100%
 librist-0.2.10-1...   100.7 KiB   547 KiB/s 00:00 [######################] 100%
 python-typing_ex...    59.9 KiB   247 KiB/s 00:00 [######################] 100%
 python-tqdm-4.66...   124.5 KiB   253 KiB/s 00:00 [######################] 100%
 ffmpeg-mpp-2.6.0...    12.1 MiB  7.73 MiB/s 00:02 [######################] 100%
 cjson-1.7.16-1-a...    25.8 KiB  59.5 KiB/s 00:00 [######################] 100%
 librga-multi-1.9...    82.7 KiB   143 KiB/s 00:01 [######################] 100%
 Total (11/11)          17.5 MiB  10.5 MiB/s 00:02 [######################] 100%
(11/11) checking keys in keyring                   [######################] 100%
(11/11) checking package integrity                 [######################] 100%
:: File /var/cache/pacman/pkg/ffmpeg-mpp-2.6.0-10-aarch64.pkg.tar.zst is corrupted (invalid or corrupted package (checksum)).
Do you want to delete it? [Y/n] Y
error: failed to commit transaction (invalid or corrupted package (checksum))
Errors occurred, no packages were upgraded.

```

7Ji, 2023-12-14:
Sorry, I made a mistake in the the package file matching pattern when changing the release model and thus the repo database was not correctly updated during the partial updates . This should have been fixed by 2b98f79c697ac960aa5805f832ef335e6bcecb90 , and I've forced a full update to the github releases so you can install without problem now.

Thank you for reporting.

closed 2023-12-14 by 7Ji

JFLim1, 2023-12-14:
Hi @7Ji,

Thank you.  Successfully installed "ffmpeg-mpp"  and "mpv" from your Repo.