7Ji/archrepo #17: After Upgrading to Latest  Chromium-mpp version 119.0.6045.199 VPU HW Acceleration No Longer Work

JFLim1, 2024-01-01:
Hi @7Ji,

Performed "sudo pacman -Syu" which upgraded chromium-mpp-114.0.5735.198 to Chromium-mpp-119.0.6045.199, vpu hardware acceleration no longer work.

Had tried to re-enable "sudo systemctl enable --now libv4l-rkmpp-setup.service", reboot but chromium-mpp- 119.0.6045.199  still does not support vpu hw acceleration.

chromium-mpp-114.0.5735.198 vpu hw acceleration works.  

HAPPY NEW YEAR.

Thank you for your great work and support.

7Ji, 2024-01-01:
Currently it does not work also on my board, sorry for the trouble. I pushed it to the repo after a successful build but before testing on actual hardware (as both are working to compile packages). It seems chromium doesn't want to load any v4l2 plugin no matter how I tries. For now, please stick with 114. If I can't get 119 to work I'll revert the package back to 114.

7Ji, 2024-01-01:
https://github.com/7Ji-PKGBUILDs/.report/issues/10

7Ji, 2024-01-01:
I've down/upgraded the one in repo to 114.0.5735.198-4, previously -3 before being bumped to 119, it has backported icu patch and built and tested to work properly.

I'm currently fixing a mainline kernel issue and chromium-mpp 119 would be held back for a while.

closed 2024-01-01 by 7Ji