7Ji/archrepo #18: mpp-git dependency on systemd for udev

SputnikRocket, 2024-01-05:
It would be desirable to replace the "systemd" dependency with "udev" this would allow non-systemd users to install the package, but also keep the needed dependencies, as `systemd` provides `udev`, while `eudev` provides `udev` on non-systemd systems.

### Associated PR:
[https://github.com/7Ji-PKGBUILDs/mpp-git/pull/2](https://github.com/7Ji-PKGBUILDs/mpp-git/pull/2)

### has this been tested to build and install successfully on both systemd and non-sytemd systems?
Yes. I have.

7Ji, 2024-01-05:
The tracking issue regarding PKGBUILDs should be opened at https://github.com/7Ji-PKGBUILDs/.report/issues , not here. This issue section is for stuffs directed related to releases packages, i.e. their binary form. But for this, let's keep it here.

This looks OK and could be merged. But could you elaborate on what environment you're / expect mpp to run in? My repo is only for Arch x86_64 and ALARM aarch64, not any other distro. So for that PR to the PKGBUILD, it's OK. But for this repo, it's kinda not.

SputnikRocket, 2024-01-05:
Ok. thank you for clarifying.

closed 2024-01-05 by SputnikRocket

7Ji, 2024-01-05:
The package uploaded by the builder a couple of minutes ago should already contains the dep change, but you'll need a manual re-installation as the pkgver and pkgrel were not updated alongside.

SputnikRocket, 2024-01-10:
> but you'll need a manual re-installation as the pkgver and pkgrel were not updated alongside.

I guess should have incremented pkgrel in the PR as well. silly me