7Ji/archrepo #19: After Upgrading to MPV -1:0.37.0-1, MPV Will Stream/Play Video

JFLim1, 2024-01-09:
Hi,

After the latest system upgrade where MPV is upgraded to  MPV -1:0.37.0-1 from ALARM Repo, MPV no longer stream Youtube or play local videos.

```
[jfl@opi5plus ~]$ mpv https://www.youtube.com/watch?v=7PIji8OubXU
mpv: symbol lookup error: mpv: undefined symbol: av_packet_side_data_get, version LIBAVCODEC_60

[jfl@opi5plus ~]$ mpv /media/Elements/GT_KING_PRO/videos/Big Buck Bunny 60fps 4K - Official Blender Foundation Short Film [aqz-KE-bpKQ].mp4
mpv: symbol lookup error: mpv: undefined symbol: av_packet_side_data_get, version LIBAVCODEC_60

[jfl@opi5plus ~]$ pacman -Qs ffmpeg
local/ffmpeg-mpp 2:6.0-10
    Complete solution to record, convert and stream audio and video supporting
    rockchip MPP hardware decoder
```

Not sure whether it is a conflict with "ffmpeg-mpp". Also notice "MPV" is no longer available on your 7Ji/archrepo.

7Ji, 2024-01-09:
@JFLim1 
> Also notice "MPV" is no longer available on your 7Ji/archrepo.

The latest mpv needs newer libplacebo which does not exist yet in the ALARM repo, I've just added it to my repo's build list but we'll need to wait a while before that comes online and mpv could be built against the latest libplacebo.
```
> cat 20240108_035155_build_mpv-7a13022475ba0f52ca8bf487c265ed2025d47d25-eba37a2574826f16.log
==> Making package: mpv 1:0.37.0-1 (Mon Jan  8 03:51:56 2024)
==> WARNING: Skipping dependency checks.
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
+ exec meson setup --prefix /usr --libexecdir lib --sbindir bin --buildtype plain --auto-features enabled --wrap-mode nodownload -D b_lto=false -D b_pie=true -D python.bytecompile=1 mpv build --auto-features auto -Dlibmpv=true -Dcaca=disabled -Dcdda=enabled -Ddvbin=enabled -Ddvdnav=enabled -Dlibarchive=enabled -Dopenal=enabled
The Meson build system
Version: 1.3.1
Source dir: /srv/repo/build/mpv/src/mpv
Build dir: /srv/repo/build/mpv/src/build
Build type: native build
Project name: mpv
Project version: 0.37.0
C compiler for the host machine: ccache cc (gcc 12.1.0 "cc (GCC) 12.1.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: aarch64
Host machine cpu: aarch64
Program python3 found: YES (/usr/bin/python3)
Found pkg-config: YES (/usr/bin/pkg-config) 2.1.0
Run-time dependency libavcodec found: YES 60.3.100
Run-time dependency libavfilter found: YES 9.3.100
Run-time dependency libavformat found: YES 60.3.100
Run-time dependency libavutil found: YES 58.2.100
Run-time dependency libswresample found: YES 4.10.100
Run-time dependency libswscale found: YES 7.1.100
Dependency libplacebo found: NO found 5.264.1 but need: '>=6.338.0'
Did not find CMake 'cmake'
Found CMake: NO
Run-time dependency libplacebo found: NO 

mpv/meson.build:26:13: ERROR: Dependency lookup for libplacebo with method 'pkgconfig' failed: Invalid version, need 'libplacebo' ['>=6.338.0'] found '5.264.1'.

A full log can be found at /srv/repo/build/mpv/src/build/meson-logs/meson-log.txt
==> ERROR: A failure occurred in build().
    Aborting...
```

@hbiyik 

Thank you for your kind work! 

> may be afterwards you can also remove the custom mpv in your repo.

Yeah that should really be done sooner or later... As you can see tracking Arch directly ahead of ALARM is yet another source of breakage.

JFLim1, 2024-01-09:
> The latest mpv needs newer libplacebo which does not exist yet in the ALARM repo,

Hi @7Ji,
Not sure I understand you correctly.

When mpv-1:0.36.0-1 was upgraded to mpv-1:0.37.0-1 from ALARM the "libplacebo-6.338.1-1-aarch64.pkg.tar.xz" was installed.

7Ji, 2024-01-09:
They updated libplacebo just now, not in the past few days when mpv was broken, and right after I added libplacebo to my build list just now....

hbiyik, 2024-01-09:
@7Ji 

done,
ffmpeg-mpp and ffmpeg4.4-mpp needs to be rebuild.
mpv should be removed from the 7ji Repo this will not work anymore, instead the one in the Alarm repo must be used.

Side note: after the recent change of mpp-git from systemd to dbus, i got an error that  dbus-units dependency could not be found, i assume this is some archlinux temporary issue.

7Ji, 2024-01-10:
ffmpeg{,-4.4}-mpp both built a couple hours ago and uploaded

mpv and libplacebo removed from the build list.

JFLim1, 2024-01-10:
> ffmpeg{,-4.4}-mpp both built a couple hours ago and uploaded

Hi @7Ji,

The ffmpeg-mpp is still named https://github.com/7Ji/archrepo/releases/download/aarch64/ffmpeg-mpp-2.6.0-10-aarch64.pkg.tar.zst -- Uploaded 7 hours ago. Is this the new ffmpeg-mpp package.

sudo pacman -Syu

Did not trigger update/upgrade "ffmpeg-mpp".  Do I have to reinstall "ffmpeg-mpp"?  Thanks.

7Ji, 2024-01-10:
Not yet, the fix to ffmpeg fork was pushed by hbiyik to his repo 6 hours ago

7Ji, 2024-01-10:
The ffmpeg-mpp PKGBUILD seems to be broken after the bump:
```
> cat 20240110_055046_build_ffmpeg-mpp-65c46d2a83c21c57056b3ab6def31aebdc46734e-e70f180e10bd5eb2-6.1.1.log
==> Making package: ffmpeg-mpp 2:6.1.1-1 (Wed Jan 10 05:50:49 2024)
==> WARNING: Skipping dependency checks.
==> WARNING: Using existing $srcdir/ tree
==> Starting pkgver()...
==> Starting build()...
ERROR: libplacebo >= 4.192.0 not found using pkg-config

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.libera.chat.
Include the log file "ffbuild/config.log" produced by configure as this will help
solve the problem.
==> ERROR: A failure occurred in build().
    Aborting...
```
I'm fixing it.

7Ji, 2024-01-10:
Now 6.1-1 was properly built and updated

closed 2024-01-10 by 7Ji