7Ji/archrepo #20: After Upgrading to MPV-1:0.37.0-1 & ffmpeg-mpp-2.6.1.1-1, MPV High Drop Frames When Streaming Video With 60FPS

JFLim1, 2024-01-11:
Hi,

Just to share and not sure whether it is a bug.

After upgrading to mpv-1:0.37.0-1 and ffmpeg-mpp-2.6.1.1-1, MPV have very high drop frames when streaming video with 60 FPS compared to  mpv to version mpv-1:0.36.0-1 and ffmpeg-mpp-2.6.0-10.  One example is "Bib Buck Bunny" video

> mpv --gpu-context=x11egl --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ

Downgrade mpv & ffmpeg-mpp to mpv-1:0.36.0-1 and ffmpeg-mpp-2.6.0-10, the same video and settings stream with low drop frames.

hbiyik, 2024-01-11:
use `mpv --hwdec=rkmpp ....` then you will have no frame drops

JFLim1, 2024-01-11:
> use `mpv --hwdec=rkmpp ....` then you will have no frame drops

Hi @hbiyik,

Tried that also.  Re-upgrade mpv to mpv-1:0.37.0-1 and ffmpeg-mpp-2:6.1.1-1.  The results is the same. Very high drop frames.

[image]

```
[jfl@opi5plus ~]$ mpv --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Using hardware decoding (rkmpp).
AO: [pipewire] 48000Hz 5.1 6ch floatp
VO: [gpu] 3840x2160 drm_prime[nv12]
Exiting... (Quit)
AV: 00:03:40 / 00:10:34 (35%) A-V:  0.000 Dropped: 6534 Cache: 65s/150MB

```

hbiyik, 2024-01-11:
This looks like more of a rendering issue, seems they have changed some pipes

you can use libmali together with gpu-next as a better option

`libmali mpv -vo=gpu-next --hwdec=rkmpp youtubeurl`

JFLim1, 2024-01-11:
> This looks like more of a rendering issue, seems they have changed some pipes
> 
> you can use libmali together with gpu-next as a better option. Panfork is not the best when it comes to rendering high res stuff. Hopefully will be better with panthor mesa.

Hi @hbiyik,

Downgraded to mpv-1:0.36.0-1 ffmpeg-mpp-2:6.0-10 and libplacebo-5.264.1-4 and the video stream with very low drop frames 

Question is it safe to continue to use the older mpv-1:0.36.0-1, ffmpeg-mpp-2:6.0-10 and libplacebo-5.264.1-4?  

```
mpv --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Using hardware decoding (rkmpp).
AO: [pipewire] 48000Hz 5.1 6ch floatp
VO: [gpu] 3840x2160 drm_prime[nv12]
AV: 00:10:34 / 00:10:34 (100%) A-V:  0.000 Dropped: 18 Cache: 0.0s

Exiting... (End of file
```)

7Ji, 2024-01-11:
> Question is it safe to continue to use the older mpv-1:0.36.0-1, ffmpeg-mpp-2:6.0-10 and libplacebo-5.264.1-4?

Keeping old package on your rolling release system is a big no go. There's a reason why this repo rebuilds every package so aggressively: even just a single dependency upgrade would possibly break your old package and you would then need to also downgrade every single dep.

hbiyik, 2024-01-11:
ok found it,

use

` mpv -vo=gpu-next --profile=fast  --hwdec=rkmpp youtubelink`

the important part is --profile=fast with mpv 0.37

JFLim1, 2024-01-11:
> ok found it,
> 
> use
> 
> ` mpv -vo=gpu-next --profile=fast --hwdec=rkmpp youtubelink`

Thank you for the tip on how to run the new mpv-0.37 command.

FYI, I did try "mpv --profile=fast --hwdec=rkmpp <youtubelink>" earlier and it did not work.

Re-upgrade/install mpv-0.37 ffmpeg-mpp-2:6.1.1-1, with the new command:

```
[jfl@opi5plus ~]$ mpv --vo=gpu-next --profile=fast --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Using hardware decoding (rkmpp).
AO: [pipewire] 48000Hz 5.1 6ch floatp
VO: [gpu-next] 3840x2160 drm_prime[nv12]
[ffmpeg/video] vp9_rkmpp_decoder: An invalid frame was output by a decoder. This is a bug, please report it.
Error while decoding frame (hardware decoding)!
Exiting... (End of file)
AV: 00:10:34 / 00:10:34 (100%) A-V:  0.000 Dropped: 27 Cache: 0.0swarning: queue 0xffff74053760 destroyed while proxies still attached:
  wl_display@1 still attached
[jfl@opi5plus ~]$ 
```
NO more high drop frames!  

Re-tried just with "mpv --profile=fast --hwdec=rkmpp <youtubelink>" and also NO more high drop frames.
`
mpv --profile=fast --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ`

Thank you for your support and great work.

hbiyik, 2024-01-11:
alo better to gpu-next it is more performant. The core of the issue is rendering related, namely panfork is lacking up somewhere. I think mpv guys changed the default profile to quaility or something so now frames are dropped. But the drop is not because of decoder but rendering. For high quality renderin gyou can sue libmali it is in the 7ji repo. Or wait for panthor to be released.

Closing now.

closed 2024-01-11 by hbiyik