7Ji/archrepo #21: Pacman update, failed retrieving file '7Ji.db'

alex-morgun, 2024-01-16:
```
pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 alarm is up to date
 aur is up to date
 7Ji.db failed to download
error: failed retrieving file '7Ji.db' from github.com : The requested URL returned error: 404
error: failed to synchronize all databases (failed to retrieve some files)
```

7Ji, 2024-01-16:
I'm doing a full update and it should come back online after ~20 minutes:

https://github.com/7Ji/archrepo/releases/tag/untagged-b36b8b120bbb11fb8058

7Ji, 2024-01-16:
https://github.com/7Ji/archrepo/releases/tag/aarch64

It's back

closed 2024-01-16 by 7Ji

7Ji, 2024-01-17:
This shouldn't happen again (or at least not as frequently) after https://github.com/7Ji/archrepo/commit/b017ef69d9b8539a7617c7a47bf31bad8e0fd5b8 as I've switched to a file-by-file update method.