7Ji/archrepo #24: Latest mpp-git r3456.0e03d3bc-1 breaks chromium-mpp and mpv vpu hw acceleration on Opi5-Plus

JFLim1, 2024-02-13:
Hi @7Ji,

For your information, the latest "mpp-git r3456.0e03d3bc-1" in the 7Ji-Archrepo breaks chromium-mpp and mpv vpu hw acceleration.  chromium-mpp have visual artifacts and does not stream 1080p or 4K/60 Youtube video -- Freezes and high CPU.

MPV also have similar issue when streaming Youtube video and report issue with "VP9....", video not smooth and high drop frames.

Revert to **"mpp-git r3456.0e03d3bc-1"** using:

`sudo pacman -U /var/cache/pacman/pkg/mpp-git-r3456.0e03d3bc-1-aarch64.pkg.tar.zst`

And both chromium-mpp and mpv streams 4K/60 video well with vpu hw acceleration.

Wish you A Happy Prosperous Chinese New Year.

hbiyik, 2024-02-13:
solution: https://github.com/hbiyik/agrrepo/commit/a67b96bf54798872b16749bec41abae66b0ef8f8

i will update when i get to hone

7Ji, 2024-02-13:
I'm creating a none-git mpp package and that would stick to 1.0.3 tag until the issue is resolved, as I don't think a -git package should stick to a frozen tag rather than a branch/HEAD.

7Ji, 2024-02-13:
```
> pacman -Ss '^mpp'
7Ji/mpp 1.0.3-1
    Rockchip VPU Media Process Platform (MPP) for hardware video decode
7Ji/mpp-git 1.0.4.r3472.fdeb8c37-1
    Rockchip VPU Media Process Platform (MPP) for hardware video decode latest revision from git
```
@JFLim1 
You can install `mpp` instead of `mpp-git`, it would be frozen to `1.0.3` until issues resolved. 

@hbiyik 
Appreciate for the quick response and help, but I think that's not needed for `mpp-git` now. I've also updated the `pkgver()` function for `mpp-git` so it's now easier to tell which tag it's based on.

closed 2024-02-13 by 7Ji

hbiyik, 2024-02-13:
i should migrate them to mpp now,