7Ji/archrepo #29: Conflict Between New ffmpeg-2:7.0.1-1 &  ffmpeg-mpp-git

JFLim1, 2024-06-30:
Not sure which package in 7Ji/archrepo is triggering the requirement to upgrade/install ffmpeg-2:7.0.1-1.

sudo pacman -Syu
[sudo] password for jfl: 
:: Synchronizing package databases...
 core is up to date
 extra                                  9.2 MiB  43.1 KiB/s 03:39 [####################################] 100%
 community is up to date
 7Ji is up to date
 rockchip is up to date
:: Starting full system upgrade...
warning: mpv: ignoring package upgrade (1:0.38.0-4 => 1:0.38.0-5)
resolving dependencies...
looking for conflicting packages...
:: ffmpeg-2:7.0.1-1 and ffmpeg-mpp-git-6.1.r112638.9efe5bcff0-1 are in conflict. Remove ffmpeg-mpp-git? [y/N] N
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: ffmpeg-2:7.0.1-1 and ffmpeg-mpp-git-6.1.r112638.9efe5bcff0-1 are in conflict

7Ji, 2024-06-30:
Install `pacman-contrib` and run `pactree -r ffmpeg`

And what's repo `rockchip`?

JFLim1, 2024-06-30:
> And what's repo `rockchip`?

"rockchip" is @samuel or kwankiu repo -- https://github.com/kwankiu/PKGBUILDs/releases/aarch64.  Testing Panthor.

> Install pacman-contrib
```
pacman -Qs pacman-contrib
local/pacman-contrib 1.10.6-1
    Contributed scripts and tools for pacman systems
```

> run pactree -r ffmpeg
```
sudo pactree -r ffmpeg
[sudo] password for jfl: 
ffmpeg-mpp-git provides ffmpeg
├─chromaprint
│ ├─grilo-plugins
│ │ └─gnome-music
│ └─gst-plugins-bad
│   ├─grilo-plugins
│   └─totem
├─ffmpegthumbnailer
├─firefox
├─gst-libav
├─mpv
│ └─celluloid
├─opencv
│ └─gst-plugin-opencv
└─scratch3-bin
```

7Ji, 2024-06-30:
At least "chromaprint" and "mpv" need newer libav{codec,device,filter,format,util}.so which is only provided by newer `ffmpeg`:
```
> pacman -Si mpv | grep libav
Depends On      : alsa-lib  libasound.so=2-64  desktop-file-utils  ffmpeg  libavcodec.so=61-64  libavdevice.so=61-64  libavfilter.so=10-64  libavformat.so=61-64  libavutil.so=59-64  libswresample.so=5-64  libswscale.so=8-64  glibc  hicolor-icon-theme  jack  libjack.so=0-64  lcms2  liblcms2.so=2-64  libarchive  libarchive.so=13-64  libass  libass.so=9-64  libbluray  libbluray.so=2-64  libcdio  libcdio-paranoia  libdrm  libdvdnav  libdvdread  libegl  libgl  libglvnd  libjpeg  libjpeg.so=8-64  libplacebo  libplacebo.so=338-64  libpulse  libpulse.so=0-64  libsixel  libva  libva.so=2-64  libva-drm.so=2-64  libva-wayland.so=2-64  libva-x11.so=2-64  libvdpau  libx11  libxext  libxkbcommon  libxkbcommon.so=0-64  libxpresent  libxrandr  libxss  libxv  luajit  mesa  mujs  libpipewire  rubberband  librubberband.so=2-64  openal  uchardet  vapoursynth  vulkan-icd-loader  wayland  zlib
```
```
> pacman -Si chromaprint | grep libav
Depends On      : gcc-libs  glibc  libavcodec.so=61-64  libavformat.so=61-64  libavutil.so=59-64  libswresample.so=5-64
```
Which is not provided by `ffmpeg-mpp-git`:
```
> pacman -Qi ffmpeg-mpp-git | grep libav
Provides        : libavcodec.so=60-64  libavdevice.so=60-64  libavfilter.so=9-64  libavformat.so=60-64  libavutil.so=58-64  libpostproc.so=57-64  libswresample.so=4-64  libswscale.so=7-64  ffmpeg=6.1.r112638.9efe5bcff0-1  ffmpeg-obs=6.1.r112638.9efe5bcff0-1  ffmpeg-rockchip  ffmpeg-rockchip-git
```
So pacman decides to replace ffmpeg-mpp-git with ffmpeg, as its the only provider for the newer .so s.

This would be the case only `ffmpeg-mpp-git` is updated. Unfortuantely it wouldn't be updated before the upstream updates: https://github.com/nyanmisaka/ffmpeg-rockchip

JFLim1, 2024-06-30:
Tried to trick the system by "ignoring mpv chromaprint" upgrade but there are other dependency issue.

```
sudo pacman -Syu
[sudo] password for jfl: 
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 7Ji is up to date
 rockchip is up to date
:: Starting full system upgrade...
warning: chromaprint: ignoring package upgrade (1.5.1-7 => 1.5.1-8)
warning: mpv: ignoring package upgrade (1:0.38.0-4 => 1:0.38.0-5)
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: installing protobuf (27.2-1) breaks dependency 'protobuf=25.3' required by python-protobuf
```

7Ji, 2024-07-01:
You'll need to find more that depends on it by yourself.

hbiyik, 2024-07-06:
should be fixed when latest ffmpeg-mpp is built with 7.x

hbiyik, 2024-07-07:
@JFLim1 i am almost sure your issues should be gone now with the latest udpates, feel free to repoen is you have further issues.

closed 2024-07-07 by hbiyik

JFLim1, 2024-07-10:
> @JFLim1 i am almost sure your issues should be gone now with the latest udpates, feel free to repoen is you have further issues.

@hbiyik Thank you.