7Ji/archrepo #30: re-enable linux-aarch64-rockchip-armbian-git?

kwankiu, 2024-07-05:
linux-aarch64-rockchip-armbian-git is now updated and fixed.

7Ji, 2024-07-05:
Closed by 3d4c37d19ab6bf77463fa07f7d094337bbc88b8f

closed 2024-07-05 by 7Ji

hbiyik, 2024-07-05:
working on the rest of the packages now, i see a lot has been broken.

kwankiu, 2024-07-13:
@7Ji btw is there any problem on building this package? Coz i noticed that the package on your repository is 6.8.11 which seems to be a really old package built two months ago, i have tested the pkgbuild myself with ARB and i am able to build the package without any issue, it is currently on 6.10-rc7.

7Ji, 2024-07-14:
> @7Ji btw is there any problem on building this package? Coz i noticed that the package on your repository is 6.8.11 which seems to be a really old package built two months ago, i have tested the [...]

Yes there is, please build in a clean chroot. It was broken ever since 6.9

```
> tail 20240713_085016_build_linux-aarch64-rockchip-armbian-git-f506024437ff233a8487bd64bfcc030e69ca3635-210acd080ed6f6ea-6.10.0.rc7.r18.a422ac.088d3b.log
arch/arm64/boot/dts/rockchip/rk3588s-khadas-edge2.dts:911.7-918.4: Warning (graph_child_address): /usb@fc000000/port: graph node has single child node 'endpoint@0', #address-cells/#size-cells are not necessary
arch/arm64/boot/dts/rockchip/rk3588s-rock-5a.dts:412.3-27: ERROR (duplicate_property_names): /spi@feb20000/pmic@0:system-power-controller: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
make[3]: *** [scripts/Makefile.lib:434: arch/arm64/boot/dts/rockchip/rk3588s-rock-5a.dtb] Error 2
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:485: arch/arm64/boot/dts/rockchip] Error 2
make[1]: *** [/srv/repo/build/linux-aarch64-rockchip-armbian-git/src/linux-stable/Makefile:1405: dtbs] Error 2
make: *** [Makefile:240: __sub-make] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
[linux-aarch64-rockchip-armbian-git-f506024437ff233a8487bd64bfcc030e69ca3635-210acd080ed6f6ea-6.10.0.rc7.r18.a422ac.088d3b.log.gz](https://github.com/user-attachments/files/16215440/linux-aarch64-rockchip-armbian-git-f506024437ff233a8487bd64bfcc030e69ca3635-210acd080ed6f6ea-6.10.0.rc7.r18.a422ac.088d3b.log.gz)

7Ji, 2024-07-14:
Ok I did a successful build after cleaning all ccache, seems my ccache was corrupted.