7Ji/archrepo #32: linux-aarch64-rockchip-bsp5.10-joshua-git needs a rebuild

kwankiu, 2024-07-18:
linux-aarch64-rockchip-bsp5.10-joshua-git is current broken as it was built from a commit that was work-in-progress after a rebase, there are newer commits that fixes the issue after that build and it currently needs a rebuild to work properly.

7Ji, 2024-07-19:
But it is at `5.10.209.r1089407.g88db26e6-1`, git commit `88db26e6`, which is the branch head of the upstream [`jammy` branch](https://github.com/Joshua-Riek/linux-rockchip/commits/jammy/), no?

```
> pacman -Si linux-aarch64-rockchip-bsp5.10-joshua-git
Repository      : 7Ji
Name            : linux-aarch64-rockchip-bsp5.10-joshua-git
Version         : 5.10.209.r1089407.g88db26e6-1
Description     : The linux-rockchip kernel, with patches picked by Joshua Riek focusing on RK3588
Architecture    : aarch64
URL             : https://github.com/Joshua-Riek
Licenses        : GPL2
Groups          : None
Provides        : None
Depends On      : coreutils  kmod  initramfs
Optional Deps   : wireless-regdb: to set the correct wireless channels of your country
Conflicts With  : None
Replaces        : None
Download Size   : 21.42 MiB
Installed Size  : 100.74 MiB
Packager        : Guoxin Pu <pugokushin@gmail.com>
Build Date      : Sun 14 Jul 2024 11:14:36 AM CST
Validated By    : SHA-256 Sum
```

kwankiu, 2024-07-19:
> But it is at `5.10.209.r1089407.g88db26e6-1`, git commit `88db26e6`, which is the branch head of the upstream [`jammy` branch](https://github.com/Joshua-Riek/linux-rockchip/commits/jammy/), no?
>
> ``` [...]

Oh right, that seems like already using the latest commit.
I have tested this kernel on radxa rock 5b, cm5, and khadas edge 2 and have no luck getting it work, there are no display output until the system reaches gdm and it stuck once you enter password. I have talked to joshua and he pointed out that commit.

kwankiu, 2024-07-20:
@7Ji @wyf9661 @hbiyik FYI,

tested joshua's ubuntu image and there's no display output issue (however experienced reduced performance in glmark2 with panfork)

on arch linux, using this package (linux-aarch64-rockchip-bsp5.10-joshua-git) with panfork, libmali, mali-g610-valhall-firmware as well as joshua firmware provied by this repository:

- radxa rock 5b: no display output at all
- radxa cm5 and khadas edge2, no display until gdm and stuck/freeze after login

7Ji, 2024-07-24:
There probably are some driver dependency issues, try to use mkinitcpio with a fallback profile or booster with `universal: true` config.

JFLim1, 2024-07-29:
Tried linux-aarch64-rockchip-bsp5.10-joshua-git on Opi5-Plus a week ago.  It just boot into a "Black or Blank Screen".  Stuck there.

JFLim1, 2024-08-01:
Hi @7Ji @kwankiu,

Upgraded to latest linux-aarch64-rockchip-bsp5.10-joshua-git-headers-5.10.209.r1089409.g897fcc76-1-aarch64.pkg.tar.zst.  It boot to a "Black Screen".

Also noticed "vmlinuz-linux-aarch64-rockchip-bsp5.10-joshua-git" is listed as "binary" instead of DOS Executable.

[image]

kwankiu, 2024-08-02:
> Hi @7Ji @kwankiu,
>
> Upgraded to latest linux-aarch64-rockchip-bsp5.10-joshua-git-headers-5.10.209.r1089409.g897fcc76-1-aarch64.pkg.tar.zst. It boot to a "Black Screen".
>
> Also noticed [...]

@7Ji @JFLim1 @w568w, seems like we are using the wrong config, 

on BSP6.1, we have updated the config location:
https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp6.1-joshua-git/commit/c400af58a0b27bb5e5f2e3122c441896f32d618f

I have applied this change to the BSP5.10, I think that would likely fix the issue.

JFLim1, 2024-08-02:
> I have [applied this change](https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp5.10-joshua-git/commit/976423a47b0ce5fe3a470453d25f831e3c7ae8bd) to the BSP5.10, I think that would likely fix the issue.

Hi @kwankiu,
Tested [linux-aarch64-rockchip-bsp5.10-joshua-git-5.10.209.r1089412.gdf2b5066-1-aarch64.pkg.tar.zst](https://github.com/7Ji/archrepo/releases/download/aarch64/linux-aarch64-rockchip-bsp5.10-joshua-git-5.10.209.r1089412.gdf2b5066-1-aarch64.pkg.tar.zst)
Still boot to "Black Screen", will test again when a new kernel linux-aarch64-rockchip-bsp5.10-joshua-git-5.10.209 with new "config" is released.

On Opi5-Plus  with RTL8852BE-CG WiFi module (by Xulong Software) stable and fast WiFi connection so far only on kernel 5.10.y.  Kernel-6.1 somehow WiFi connection is not that stable or fast enough.

kwankiu, 2024-08-02:
> Hi @kwankiu, Tested [...]

@7Ji can you rebuild this package?

7Ji, 2024-08-02:
The one on the releaae page is already built from latest commit
976423a47b0ce5fe3a470453d25f831e3c7ae8bd

kwankiu, 2024-08-02:
> […](#)

Thanks, i will test it out, the vmlinuz looks normal from the latest build.
I thought it wasnt updated as the package was uploaded less than an hour after I apply this change.

Btw, just curious, are you currently building on OPI5 or on Oracle Cloud (4 OCPU 24G RAM?)? How long does it takes you to build this kernel?

kwankiu, 2024-08-02:
@JFLim1, tested the latest 5.10 on the Khadas Edge 2 and it works. Make sure you have the correct boot config, mali firmware and panfork.
[image]

closed 2024-08-02 by kwankiu

JFLim1, 2024-08-03:
@7Ji  @kwankiu,

The new rebuild linux-aarch64-rockchip-bsp5.10-joshua-git 5.10.209.r1089412.gdf2b5066-1 boot successfully.  Thank you.