7Ji/archrepo #34: Kodi doesn't restart/quit/exit

Danibos, 2024-08-10:
Hi!

after installing the Backup addon, Kodi is unable to restart, quit, or exit. It just restart Kodi

A Kodi crash logs is generated every time, but I cannot identify the issue

Are you aware of this issue?

Please let me know how can i help with this
 
[kodi_crashlog-20240810_003943.log](https://github.com/user-attachments/files/16569608/kodi_crashlog-20240810_003943.log)

Opi5 with 4 gb ram, running kwankiu dist

pkg is kodi-mpp-git - r175807.8adee68acd-1

Thanks!

hbiyik, 2024-08-10:
This is actually a python bug,
https://github.com/python/cpython/pull/118618#issuecomment-2226450752

Before doing anything i would wait for tje next python version

Danibos, 2024-08-11:
> This is actually a python bug,
> https://github.com/python/cpython/pull/118618#issuecomment-2226450752
> 
> Before doing anything i would wait for tje next python version

Thanks for the feedback

hbiyik, 2024-08-12:
closing then

closed 2024-08-12 by hbiyik