7Ji/archrepo #36: pacman 7.0 requires the kernel supports landlock, linux-aarch64-rockchip-bsp6.1-joshua-git 6.1.75

endeavouros-arm, 2024-10-08:
Archlinux pacman 7.0 has added support for downloading packages as a separate user with dropped privileges.
https://archlinux.org/news/manual-intervention-for-pacman-700-and-local-repositories-required/

This requires that the kernel supports landlock.  Running with linux-aarch64-rockchip-bsp6.1-joshua-git 6.1.75 yields the following error.

error: restricting filesystem access failed because landlock is not supported by the kernel!

This doesn't stop pacman from updating, it just isn't utilizing the new security feature, which would be nice to have available.

I don't know if this does you any good, but adding the following to the PKGBUILD config file added landlock to kernels I compile for Odroid N2 and Pinebook Pro.

change from
CONFIG_SECURITY_LANDLOCK is not set
to
CONFIG_SECURITY_LANDLOCK=y

and from
CONFIG_LSM="yama,loadpin,safesetid,integrity,selinux,smack,tomoyo,apparmor"
to
 CONFIG_LSM="yama,loadpin,safesetid,integrity,selinux,smack,tomoyo,apparmor,landlock"

For me, this added landlock to the kernel and the new feature works.
In an existing install, pacman will generate some .pacnew files that need to be dealt with, and a new user 'alpm' is created for support for downloading packages as a separate user with dropped privileges.  

And thank you for your wonderful Archrepo, it is greatly appreciated.

hbiyik, 2024-10-08:
@endeavouros-arm the config is reused from joshua-rieks config, i would suggest to conduct the discission there?

endeavouros-arm, 2024-10-08:
What would be the best way to contact joshua?
Do you have a link to his github site?

hbiyik, 2024-10-08:
https://github.com/Joshua-Riek/linux-rockchip

He is active in the issues, so creating an issue thwre should do.

endeavouros-arm, 2024-10-09:
@hbiyik 
Thank you for your assistance.  I created an issue there.

7Ji, 2024-10-09:
I don't quite want to enable landlock on these BSP kernel packages, as they mainly track whatever the upstream provides and stays the same experience users expect from upstream, so -orangepi kernels are built from the same source + config as orange pi's kernel and -radxa from radxa, -joshua from Joshua's and so on. The only exception is -joshua which also has partially local panthor patchset but that's another story. If we really are introducing local changes then the problem comes to why would we need all these BSP kernels anyway: we would just maintain one with common changes, instead of many with a single change that needs to be de-conflicted on each update.

And even after that, you don't even know if landlock works at all on all these BSP kernels, many of them lagging behind a lot of major kernel versions. My attitude as always: test the change requested **on actual hardware**, every single one of them, that also applies for PRs. I don't want anyone updating with `pacman -Syu` ends up with a non-bootable or non-working board. 

Even if hbiyik suggested, I don't think it's a reasonable request for Joshua from Arch users to enable landlock in their Ubuntu kernel, as that brings virtually no features to Ubuntu users: https://github.com/Joshua-Riek/linux-rockchip/issues/31

That being said, I'm open to add configs to mainline or partially mainline kernel packages, like -7ji which already has landlock enabled [here](https://github.com/7Ji-PKGBUILDs/linux-aarch64-7ji/commit/e3b360def29f1f9777a97618bf2627a020426f15). You can check around and add that to others kernels with a locally vendored config.

closed 2024-10-12 by 7Ji