7Ji/archrepo #37: Fydetab Support - Please.

DarkevilPT, 2024-10-11:
Hi 7Ji
Can you please build and add `Linux-aarch64-rockchip-bsp6.1-fydetab-git` in GitHub.com/7ji/archrepo

https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp6.1-fydetab-git

7Ji, 2024-10-12:
Someone needs to build that PKGBUILD and verify it on actual hardware before adding it to the repo

kwankiu, 2024-10-13:
> Someone needs to build that PKGBUILD and verify it on actual hardware before adding it to the repo

This kernel is a fork of Joshua's noble (bsp6.1) kernel modified for the fydetab, and it is developed and tested on arch / bredOS, the major difference is that i made this PKGBUILD based on our joshua 6.1 pkgbuild using the cleaner fydetab kernel branch without panthor support commits and armbian's commits, it instead follows the structure we have with the current joshua 6.1 kernel (original kernel branch + boogie's panthor patch), without bredos's naming and dependencies issues

7Ji, 2024-10-14:
https://github.com/7Ji-PKGBUILDs/.meta/blob/main/PackageGuidelines.md#kernel-package-guidelines

If anything, fydetab is not an upstream name and it shall not be in the part for upstream. Devices name comes right after vendor name. And make it clear this is from BredOS/panda's upstream. Don't make people go to FydeOS upstream to complain for stuffs they didn't do.

kwankiu, 2024-10-14:
> https://github.com/7Ji-PKGBUILDs/.meta/blob/main/PackageGuidelines.md#kernel-package-guidelines
>
> If anything, fydetab is not an upstream name and it shall not be in the part for upstream. Devices [...]

It might sounds confusing but this is both from bredos/panda and FydeTab official 

panda worked on the kernel for bredos and got hired by fydeOS, the current pkgbuild points to a github organisation created by fydeOS officially via panda (as the maintainer)

DarkevilPT, 2024-10-14:
If only we could chat with you on discord... 7ji.. guess u would be swarmed by another hundreds of requests making u crazy.. but yeah Samuel and I are trying to expanding the fydeos tablet horizons and Panda from BredOS helped starting it.

7Ji, 2024-10-14:
https://github.com/7Ji/archrepo/commit/ba75315a878679ed1b55eee0bade72545e726f2f

closed 2024-10-14 by 7Ji