7Ji/archrepo #38: update issues

jjgalvez, 2024-10-21:
trying to update and getting this error:
error: failed retrieving file 'linux-aarch64-rockchip-bsp6.1-joshua-git-6.1.75.r1272339.g5c434126-1-aarch64.pkg.tar.zst' from github.com : Maximum file size exceeded

Hope this is something easy to remedy. 
Thanks

7Ji, 2024-10-21:
The disk space was exhausted on my building machine (took all my 256G NVMe), and during the time window between it went full and I cleaned to free up disk space and several "update" was done, so the DB was then broken due to it being "updated" in a full disk and containing corrupted data.

I've forced a full rebuild and the DB is now correct, you should be able to download correctly after my local repo is synced fullly to Github.

closed 2024-10-21 by 7Ji

jjgalvez, 2024-10-21:
Thank you,  appreciate you maintaining this for the community

hbiyik, 2024-10-21:
@7Ji you might consider opening a sponsorship section in github, a lot of people are using your service and i think some might be willing to contribute financially.

7Ji, 2024-10-22:
> @7Ji you might consider opening a sponsorship section in github, a lot of people are using your service and i think some might be willing to contribute financially.

Thanks, but Github sponsorship does not support China and I have a well-paid full-time job working on OS, kernel and compiler to support my own FOSS expenses. As always I like the feeling of doing whatever I want and can pick or drop things as I like, and not needing to say sorry to anyone if I one day decided to put down FOSS and spend more time on my job or with my family.