7Ji/archrepo #43: Enable linux-aarch64-rockchip-armbian-git?

kwankiu, 2025-02-07:
I just updated [linux-aarch64-rockchip-armbian-git](https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-armbian-git) to the latest 6.13 armbian patches and config, rk3588 patches are now merged to rockchip64, which means this kernel has support for all aarch64 rockchip SBCs supported by armbian. It should now build fine.

[linux-firmware-armbian-git](https://github.com/7Ji-PKGBUILDs/linux-firmware-armbian-git) provides most wireless drivers among popular SBCs so it is ideally to build this package as well.

Btw, [linux-aarch64-rockchip-joshua-git](https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-joshua-git) is stall as you may know Joshua has terminated his project, and a 6.11 kernel with limited rockchip support might not be useful, since we have `linux-aarch64-rockchip-armbian-git` should we archive `linux-aarch64-rockchip-joshua-git`?

FYI @7Ji @hbiyik

hbiyik, 2025-02-07:
makes sense to me do you need a review of the PKGBUILD? i can give one @kwankiu

hbiyik, 2025-02-07:
I dont really understand why the config is device specific (edge..)? Idk how armbian manages the configs but for me it does not make sense to keep a config per device.

https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-armbian-git/blob/4b28f97580e5761d1ceeb0465da49d52e2e212f2/PKGBUILD#L30

hbiyik, 2025-02-07:
So this -edge thing is not khadas edge, but edge config...

7Ji, 2025-02-08:
It fails to build

```
│drivers/gpu/drm/panel/panel-simple-dsi.c: In function 'panel_simple_get_modes':                                                                                                                                                                                                                                                                            │
│drivers/gpu/drm/panel/panel-simple-dsi.c:310:73: error: passing argument 3 of 'of_get_drm_display_mode' makes pointer from integer without a cast [-Wint-conversion]                                                                                                                                                                                       │
│  310 |         ret = of_get_drm_display_mode(panel->dev->of_node, mode, p->desc->bus_format,                                                                                                                                                                                                                                                              │
│      |                                                                  ~~~~~~~^~~~~~~~~~~~                                                                                                                                                                                                                                                               │
│      |                                                                         |                                                                                                                                                                                                                                                                          │
│      |                                                                         u32 {aka unsigned int}                                                                                                                                                                                                                                                     │
│In file included from ./include/drm/drm_crtc.h:32,                                                                                                                                                                                                                                                                                                         │
│                 from drivers/gpu/drm/panel/panel-simple-dsi.c:13:                                                                                                                                                                                                                                                                                         │
│./include/drm/drm_modes.h:511:66: note: expected 'u32 *' {aka 'unsigned int *'} but argument is of type 'u32' {aka 'unsigned int'}                                                                                                                                                                                                                         │
│  511 |                             struct drm_display_mode *dmode, u32 *bus_flags,                                                                                                                                                                                                                                                                        │
│      |                                                             ~~~~~^~~~~~~~~                                                                                                                                                                                                                                                                         │
│drivers/gpu/drm/panel/panel-simple-dsi.c: In function 'panel_simple_disable':                                                                                                                                                                                                                                                                              │
│drivers/gpu/drm/panel/panel-simple-dsi.c:379:13: warning: unused variable 'err' [-Wunused-variable]                                                                                                                                                                                                                                                        │
│  379 |         int err = 0;                                                                                                                                                                                                                                                                                                                               │
│      |             ^~~                                                                                                                                                                                                                                                                                                                                    │
│drivers/gpu/drm/panel/panel-simple-dsi.c: In function 'panel_simple_enable':                                                                                                                                                       [... cut ...]                                                                                                  │
│make[3]: *** [scripts/Makefile.build:440: drivers/gpu] Error 2                                                                                                                                                                                                                                                                                             │
│make[3]: *** Waiting for unfinished jobs....  
```

7Ji, 2025-02-08:
Make sure a package is buildable before you want to enable it. Post a full successful build log before you open this.

closed 2025-02-08 by 7Ji

kwankiu, 2025-02-28:
@7Ji, I just reverted to 6.12.y (using the armbian rockchip64 current instead of edge), and i can confirm this builds successfully.

Will review armbian rockchip64 edge later when 6.14.y is available.

```
==> Finished making: linux-aarch64-rockchip-armbian-git 6.12.17.r188.d725f1.a246a0-1 (Fri 28 Feb 2025 05:39:16 AM HKT)
```

kwankiu, 2025-03-10:
> [@7Ji](https://github.com/7Ji), I just reverted to 6.12.y (using the armbian rockchip64 current instead of edge), and i can confirm this builds successfully.
>
> Will review armbian rockchip64 edge [...]

@7Ji , this package is now updated to the latest mainline, tested build successfully:

https://github.com/1usOS/test-repo/actions/runs/13772799759/job/38515386772

7Ji, 2025-03-31:
```
arch/arm64/boot/dts/rockchip/rk3308-rock-s0.dts:19.7-30.4: ERROR (phandle_references): /leds: Reference to non-existent node or label "green_led_gio"

arch/arm64/boot/dts/rockchip/rk3308-rock-s0.dts:109.31-120.4: ERROR (phandle_references): /regulator-3v3-vcc-sd: Reference to non-existent node or label "sdmmc_2030"

arch/arm64/boot/dts/rockchip/rk3308-rock-s0.dts:132.25-141.4: ERROR (phandle_references): /vcc5v0-otg: Reference to non-existent node or label "otg_vbus_drv"

ERROR: Input tree has errors, aborting (use -f to force output)
```

kwankiu, 2025-03-31:
> ```
> arch/arm64/boot/dts/rockchip/rk3308-rock-s0.dts:19.7-30.4: ERROR (phandle_references): /leds: Reference to non-existent node or label "green_led_gio" [...]

It is now taking the latest linux kernel from the master branch, if anything merged to the latest mainline kernel it would requires the armbian side to update the patch accordingly

Armbian has a “last-known-good-tag”, one way to ensure it builds would be to first fetch that variable before cloning from source, then download from tag archive (mainline) instead of repo

Alternatively, we could create a non-git version of this pkgbuild like `linux-aarch64`, use kernel archive, patch and config at exact commit / version

kwankiu, 2025-03-31:
For now, master branch seems to have started merging 6.15 stuff while Armbian’s patch is compatible with the latest 6.14